valentinbreiz/nativeaot-patcher #6: Liquip.Patcher - Patch assembly plugs

valentinbreiz, 2024-12-17:


valentinbreiz, 2025-01-22:
Blocked by #22

kumja1, 2025-01-25:
Fixed by #25 

> Blocked by [#22](https://github.com/valentinbreiz/nativeaot-patcher/issues/22)

kumja1, 2025-03-23:
Resolved by #35

valentinbreiz, 2025-04-03:
Now we have the plug patcher + a kernel building, we should start focusing on this. We could validate the implementation with some basic debug stubs to send text to COM ports

https://github.com/valentinbreiz/nativeaot-patcher/issues/34

ascpixi, 2025-04-04:
How do assembly plugs look right now? Is there an example?

valentinbreiz, 2025-04-04:
> How do assembly plugs look right now? Is there an example? 

Only implemented things are https://github.com/valentinbreiz/nativeaot-patcher/tree/main/src/Cosmos.Asm.Build/Tasks
We need to link this with the kernel

valentinbreiz, 2025-04-04:
Blocked by #56

zarlo, 2025-04-05:
> > How do assembly plugs look right now? Is there an example?
> 
> Only implemented things are [`main`/src/Cosmos.Asm.Build/Tasks](https://github.com/valentinbreiz/nativeaot-patcher/tree/main/src/Cosmos.Asm.Build/Tasks) We need to link this with the kernel ([#56](https://github.com/valentinbreiz/nativeaot-patcher/issues/56) should be done before)

i have no clue if they work as i did not have time to test it

valentinbreiz, 2025-04-07:
Unblocked with #56 #55

closed 2025-04-09 by valentinbreiz