valentinbreiz/nativeaot-patcher #57: Fix iso-tests action (missing ilc)

valentinbreiz, 2025-04-03:
https://github.com/valentinbreiz/nativeaot-patcher/actions/runs/14250541817/job/39941990087#step:9:5471

Guillermo-Santos, 2025-04-03:
I have seen that sometimes the `DependsOnTargets` property is not respected and run without it.

[https://github.com/valentinbreiz/nativeaot-patcher/blob/7a43505ea36128b0449a9120bd46899d3def76a1/src/Cosmos.ilc.Build/ILC.Build.targets#L91](https://github.com/valentinbreiz/nativeaot-patcher/blob/7a43505ea36128b0449a9120bd46899d3def76a1/src/Cosmos.ilc.Build/ILC.Build.targets#L91)

Guillermo-Santos, 2025-04-04:
I see the test also install the ilc workload. That means that we could make something like this
```xml
<IlcToolsPath Condition="$(IlcHostPackagePath) != ''">$(IlcHostPackagePath)/tools/</IlcToolsPath>
```
And change the exec to
```xml
    <Exec
      Command="&quot;$(IlcToolsPath)ilc&quot; @&quot;%(ManagedBinary.RspFile)&quot;"
      WorkingDirectory="$(FullIntermediateOutputPath)"
    />
```

Removed an `/` on the call, since we are already adding it on the `IlcToolsPath` definition.

So, if the target did not found it, the workload should do it.

closed 2025-04-04 by valentinbreiz