valentinbreiz/nativeaot-patcher #58: Create a patched dotnet runtime (ilc systemmodule)

valentinbreiz, 2025-04-03:


valentinbreiz, 2025-04-09:
Blocked by https://github.com/valentinbreiz/nativeaot-patcher/issues/78

valentinbreiz, 2025-04-11:
Blocked by #81

zarlo, 2025-08-20:
for this are we thinking of pre patching it or doing it at build time?

valentinbreiz, 2025-08-20:
> for this are we thinking of pre patching it or doing it at build time?

Good question, I think this should be talked about on https://github.com/valentinbreiz/nativeaot-patcher/issues/92
But my idea was to include a git submodule from runtime, then use the official NativeAOT System.Private.CoreLib full BCL (https://github.com/dotnet/runtime/tree/main/src/coreclr/nativeaot/System.Private.CoreLib/src) + edit in this BCL what we need for baremetal in overriden sources (I don't remember exactly how but we can do that in csproj)

closed 2025-08-20 by valentinbreiz