valentinbreiz/nativeaot-patcher #59: Port build pipeline to windows

valentinbreiz, 2025-04-03:
Linker targets, ilc

ascpixi, 2025-04-04:
I would propose that we do everything from ilc - namely, dotnet would invoke a bash script for the native compilation step. The tooling available for Windows to manipulate the formats used by System V are subpar, and we can't avoid WSL due to ilc not supporting cross-platform compilation (and thus, we can't compile for the `linux-x64` target.

ascpixi, 2025-04-04:
Do also note, that dotnet needs to be invokable from Windows, since IDEs rely on it for intellisense and whatnot. So, we can't simply tell the user to always call dotnet through WSL. That'd also make for poor DX for Windows end-users.

zarlo, 2025-04-04:
> Do also note, that dotnet needs to be invokable from Windows, since IDEs rely on it for intellisense and whatnot. So, we can't simply tell the user to always call dotnet through WSL. That'd also make for poor DX for Windows end-users.

a lot of IDE's allow running it in WSL. so this might work

valentinbreiz, 2025-04-04:
> I would propose that we do everything from WSL - namely, dotnet would invoke a bash script for the native compilation step. The tooling available for Windows to manipulate the formats used by System V are subpar, and we can't avoid WSL due to ilc not supporting cross-platform compilation (and thus, we can't compile for the `linux-x64` target.

if we can link this to visual studio community yes

valentinbreiz, 2025-04-04:
or maybe it's time to port officially visual studio code

kumja1, 2025-04-04:
> I would propose that we do everything from WSL - namely, dotnet would invoke a bash script for the native compilation step. The tooling available for Windows to manipulate the formats used by System [...]

what about LIEF (Library to Instrument Executable Formats)

kumja1, 2025-04-04:
> I would propose that we do everything from WSL - namely, dotnet would invoke a bash script for the native compilation step. The tooling available for Windows to manipulate the formats used by System [...]

What about LIEF (Library to Instrument Executable Formats) or objconv

valentinbreiz, 2025-05-16:
Update: Works until ISO build

AzureianGH, 2025-08-24:
Build ISO now works on windows in latest.