valentinbreiz/nativeaot-patcher #60: Assembly plugs for debugging

valentinbreiz, 2025-04-03:
First send text via COM but this will be the start for the baremetal nativeaot debug engine

valentinbreiz, 2025-04-03:
Blocked by #6

ascpixi, 2025-04-04:
For the Assembly part, all we need is one function for x86-64: `x86_outb`. If we need our kernel to communicate with the debugger as well, we also need `x86_inb`.

valentinbreiz, 2025-04-04:
> For the Assembly part, all we need is one function for x86-64: `x86_outb`. If we need our kernel to communicate with the debugger as well, we also need `x86_inb`.

we are under x64 but you're right

9xbt, 2025-04-04:
> For the Assembly part, all we need is one function for x86-64: `x86_outb`. If we need our kernel to communicate with the debugger as well, we also need `x86_inb`.

x86_64 is the only modern ISA that still has port I/O, so I'd name those `outb` and `inb` respectively. Or `outportb` and `inportb` if you wanna be fancy

ascpixi, 2025-04-04:
Ideally, I'd like to adopt the naming convention of `<arch identifier>_<function name>` for *all* architecture-specific functions. That is, even for functions that only make sense for one architecture, we still should prefix it with the given identifier - as a form of namespacing.

closed 2025-04-09 by valentinbreiz