valentinbreiz/nativeaot-patcher #81: Automatically generate native stubs

valentinbreiz, 2025-04-11:
Scan native calls + generate plugs

valentinbreiz, 2025-04-11:
Blocked by #71

ascpixi, 2025-04-12:
What do we mean here by "generate"? How would we automatically implement something like `RhpAssignRef`?

valentinbreiz, 2025-04-12:
> What do we mean here by "generate"? How would we automatically implement something like `RhpAssignRef`?

It's not about the implementation it's just empty stubs. Like when IL2CPU saying missing plugs

ascpixi, 2025-04-12:
Might be better to leave them un-implemented. If `ilc` emits code that would somehow call an unimplemented native function, the linker would complain about a missing symbol. It's better to get such diagnostics at compile-time rather than runtime.

With our plug system, we should remove every method that calls an unimplemented `extern` method, and any other methods that call it, completely traversing the call graph. This would mean that we never, *ever*, may call an unimplemented method "by mistake". And if we are completely sure that we want to leave a native method unimplemented, but leave the callers intact, we would explicitly do a `throw new NotImplementedException()`.

The biggest benefit of this, is that we avoid the guessing game we had with gen2, where the end-user wasn't entirely sure what code was plugged, and what wasn't. With this approach, unplugged code doesn't even show up in IntelliSense, since the members don't exist in the metadata in the first place.

closed 2025-04-20 by kumja1