valentinbreiz/nativeaot-patcher #136: Pack Cosmos.Patcher fails in CI

valentinbreiz, 2025-08-24:
https://github.com/valentinbreiz/nativeaot-patcher/actions/runs/17190908681/job/48766401454

Guillermo-Santos, 2025-08-25:
the slnx file is missing xd

valentinbreiz, 2025-08-25:
> the slnx file is missing xd

now it's about Cosmos.Kernel.Native.x86

`/home/runner/work/nativeaot-patcher/nativeaot-patcher/examples/KernelExample/KernelExample.csproj : error NU1101: Unable to find package Cosmos.Kernel.Native.x86. No packages exist with this id in source(s): nuget.org [/home/runner/work/nativeaot-patcher/nativeaot-patcher/nativeaot-patcher.slnx]`

Guillermo-Santos, 2025-08-25:
> now it's about Cosmos.Kernel.Native.x86
>
> `/home/runner/work/nativeaot-patcher/nativeaot-patcher/examples/KernelExample/KernelExample.csproj : error NU1101: Unable to find package [...]

We need a separate solution for the packages, the issue is on the kernel example, but we shouldn't pack the kernel example

closed 2025-08-26 by valentinbreiz